EmileSpecialProducts/UltraWiFiDuck #26: api run?cmd=write returns Unknown Command

justusiv, 2025-04-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Describe your problem

Running the examples from the PowershellTestScript.ps1 i am getting this error.

### Steps to reproduce

The other API calls seem to work although i have not tested them all 

### What hardware are you using?

ESP32-S3

justusiv, 2025-04-18:
ok i think i figured out it actually needs the run?cmd=create but now there is a new issue. Whatever is passed is not in the file. its always showing up empty

EmileSpecialProducts, 2025-04-18:
Yes that is true, 
I removed this as I removed the websocket and added the upload of binary files, 
Using the function ESPSPIFFSuploadfile() also in the [PowershellTestScript.ps1](https://github.com/EmileSpecialProducts/UltraWiFiDuck/blob/master/test/PowershellTestScript.ps1)
And I added a file upload in the GUI .

justusiv, 2025-04-18:
What i like about the write command(if it worked) is you can easily run it in a one line command with powershell or using curl. While i got the ESPSPIFFSuploadfile command to work it would be really clunky to use from my phone for example. I notice the run?cmd create command actually creates a blank file. Is there anyway to modify the contents with a simple run?cmd command

justusiv, 2025-04-18:
I got it working with this. 
curl -X POST -F "data=$(printf "GUI r\nnotepad\nHello");filename=test123.txt;type=application/octet-stream" http://UltraWiFiDuck.local/upload

EmileSpecialProducts, 2025-04-21:
I will have a look to add the Write command.
And also add a Execute command that will execute a multiline string.
And update the documentation to add some CURL examples.

EmileSpecialProducts, 2025-08-21:
I have added som [CURL](https://github.com/EmileSpecialProducts/UltraWiFiDuck/blob/master/test/curlTestScript.sh) examples.

closed 2025-08-21 by EmileSpecialProducts